elephantrobotics/pymycobot #3: Python process dies when calling get_radians()

nisshan-x, 2021-01-26:
Atom : v2.4
pymycobot : v.2.1.2

Python process dies when calling get_radians(). For example, when I run test.py in your repository, the process dies.

`movement status: 1
::get_radians()
Traceback (most recent call last):
  File "./test.py", line 115, in <module>
    test(mycobot)
  File "./test.py", line 70, in test
    print('==> radians: {}\n'.format(mycobot.get_radians()))
  File "/home/nish/.local/lib/python2.7/site-packages/pymycobot/mycobot.py", line 163, in get_radians
    angles = self.__mesg(Command.GET_ANGLES, has_reply=True)
  File "/home/nish/.local/lib/python2.7/site-packages/pymycobot/mycobot.py", line 94, in __mesg
    res = self._process_recived(data, genre)
  File "/home/nish/.local/lib/python2.7/site-packages/pymycobot/common.py", line 135, in _process_recived
    res.append(self._decode_int16(a))
  File "/home/nish/.local/lib/python2.7/site-packages/pymycobot/common.py", line 63, in _decode_int16
    return struct.unpack('>h', data)[0]
struct.error: unpack requires a string argument of length 2
`

zlj-zz, 2021-01-26:
Today I tested it. It's OK. Can you provide more detailed information?

nisshan-x, 2021-01-28:
What info do you need? So far, info in my first comment is everything what I can give.

zlj-zz, 2021-01-28:
Is it still wrong now? How about `get_angles()`?

Through the information you give, the place where the error happened is incredible. If it's wrong, it should be wrong at a higher level, not here.

nisshan-x, 2021-01-28:
Ok, I will check get_angles() and give feedback to you. Hold on a moment.

nisshan-x, 2021-01-28:
Both get_radians() and get_angles() work well! I guess root cause might be instability of USB connection. The issue is solved.

Thank you!

zlj-zz, 2021-01-28:
That's OK.

closed 2021-01-28 by zlj-zz