elephantrobotics/pymycobot #7: dh parameters

mfkenson, 2021-02-10:
May I have the dh parameters of the arm?
I would like to work on trajectory generation.

zlj-zz, 2021-02-12:
what's the dh parameters?

karaage0703, 2021-02-14:
dh parameters is "Denavit-Hartenberg parameters".
For example, we can make original myCobot simulator from scratch by using dh parameters.

I made myCobot simulator.
https://github.com/karaage0703/mycobot_pi

My understanding, myCobot dh parameters is similar to UR5([reference](https://www.slideshare.net/RyanKeating13/ur5-ik))

myCobot dh parameters which I estimate is below.
https://docs.google.com/presentation/d/17b-rVN2hMOyrtATXlfU7OeVewjkEbSOUjrN6cOPCY6U/

This is Not official information. I want to know official myCobot dh parameters, too.

zlj-zz, 2021-02-14:
[image]
Maybe you can refer to this. More details need to wait.

closed 2021-02-26 by zlj-zz

kotai2003, 2021-04-04:
> dh parameters is "Denavit-Hartenberg parameters".
> For example, we can make original myCobot simulator from scratch by using dh parameters.
>
> I made myCobot simulator. [...]

Hi,
 Is your DH parameter correct? 
 The structure of robot in your figure is much different with that of myCobot. 
You need to add description how to read your DH parameter.

kotai2003, 2021-04-04:
> dh parameters is "Denavit-Hartenberg parameters".
> For example, we can make original myCobot simulator from scratch by using dh parameters.
>
> I made myCobot simulator. [...]

Hello. 
Let me ask a question on your simulator. 
In your code, you define the DH parameter as follows.

`# myCobot DH parameters
# [theta, alpha, a, d]
mycobot_sim = NLinkArm([[0., math.pi / 2, 0, 0.13156],
                    [0., 0., -0.1104, 0.],
                    [0., 0., -0.096, 0.],
                    [0., math.pi / 2, 0., 0.06639],
                    [0., -math.pi / 2, 0., 0.07318],
                    [0., 0., 0., 0.0436]])
`
I am wondering why you put zeros in theta in DH parameter?
Is there any special technique to update theta by setting up zeros initially?

Thanks.

karaage0703, 2021-04-05:
@kotai2003 
Hi, thanks for your comment.
To tell the truth, I'm not familiar with DH parameter.
I made my software according to real robot and basic robotics text.
Please refer to official information. And if you can fix my mistake, please give me Pull Request.
Sorry for I can't answer your question directly.