elephantrobotics/pymycobot #8: The mycobot adaptive gripper is always closed and gets stuck with set_gripper_value()

3110, 2021-02-21:
**Describe the bug**
Both set_gripper_value(0, speed) and set_gripper_value(496, speed) close the mycobot adaptive gripper, then the gripper gets stuck. It seems to be occured for any value of set_gripper_value().

**To Reproduce**
1. Initialize the gripper once
2. Connect to the gripper with mycobot = MyCobot("COM13")
3. Call mycobot.set_gripper_value(0, 100)  # minimum value
4. The gripper is closed and gets stuck ...
5. Power off and on
6. Connect to the gripper with mycobot = MyCobot("COM13")
7. Call mycobot.set_gripper_value(496, 100) #maximum value
8. The gripper is closed and gets stack ...

**Expected behavior**
Though I don't know the meaning of the value of set_gripper_value(), I expect the following behavior:
* The gripper is closed to maximum with set_gripper_value(0, 100).
* The gripper is opened to maximum with set_gripper_value(496, 100).

**Desktop**
 - Windows 10 Home
 - Python 3.9.2
 - pymycobot 2.2.1

**MyCobot**

- AtomMain 2.5
- MyCobot Adaptive Gripper (white)

**Additional context**
If set_gripper_ini() sets the initial position to 248, shoud be the range of the value in set_gripper_value() from 0 to 248 ?

zlj-zz, 2021-02-23:
0-496 is 0 degrees to 360 degrees.So you need to try a smaller value, like 280

3110, 2021-02-24:
Umm, the gripper is closed and gets stuck with 280(or any smaller values).  Does set_gripper_value() set the absolute position(degree?) regardless of the zero position of get_gripper_ini(), or set the relative position from the zero position of get_gripper_ini()(value = 248) ?

zlj-zz, 2021-02-24:
relative position

3110, 2021-02-24:
The values between 840(close)-2550(open) are OK...

zlj-zz, 2021-02-24:
Are you sure is 840？ not 84？

3110, 2021-02-24:
840, not 84.

3110, 2021-03-06:
With AtomMain 2.6, set_gripper_value() doesn't work at all.

zlj-zz, 2021-03-06:
I will test it next week, and response you

zlj-zz, 2021-03-12:
This method really can't be used in atom 2.6. If you have such requirements, you can use atom 2.5 first.

closed 2021-03-22 by zlj-zz