elephantrobotics/pymycobot #10: get_joint_[min|max]_value() returns wrong value for all joints

3110, 2021-02-24:
**Describe the bug**
get_joint_[min|max]_value() returns wrong value for all joints.

**To Reproduce**

```
$ ipython
Python 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.20.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from pymycobot.mycobot import MyCobot
In [2]: from pymycobot.genre import Angle
In [3]: for angle in Angle:
   ...:     print(angle.name, mycobot.get_joint_min_angle(angle.value), mycobot.get_joint_max_angle(angle.value))
J1 -15.5 15.5
J2 -16.5 16.5
J3 -16.5 16.5
J4 -16.5 16.5
J5 -17.5 17.5
J6 175.0 -216.08
```

**Expected behavior**

Acconding to User Manual V 2021.02.04(p.13), the return values shoud be the following: 
```
J1 -165.0 165.0
J2 -165.0 165.0
J3 -165.0 165.0
J4 -165.0 165.0
J5 -165.0 165.0
J6 -175.0 175.0
```
**Desktop (please complete the following information):**
- Windows 10 Home
- Python 3.9.2
- pymycobot 2.2.1

**MyCobot**
- AtomMain 2.5
- MyCobot Adaptive Gripper (white)

**Additional context**
MyCobot has been calibrated with the Calibration app in myStudio v0.0.5.

hansenmaster, 2021-03-02:
Hi Saito-san, I have the same problem. My joint min max value returns the exact value with yours.
Whenever I send command of write angle, for example angle 10 degree to J1, it moves more than 10 degree. 
Have you solved this problem?

3110, 2021-03-02:
Hi hansenmaster-san, which version of AtomMain and pymycobot are you using?  If you want to send angles with degree, you should use AtomMain 2.5 and pymycobot 2.2.0.  The AtomMain 2.3 or earlier handles angles of APIs with radian.

hansenmaster, 2021-03-03:
Hi Saito-san, thank you very much for the answer. Updating the Atom firmware fix the angle unit.

zlj-zz, 2021-03-04:
the `get_joint_min_angle` and the max has some problem, you should wait we solve this

zlj-zz, 2021-03-12:
When this problem is fixed, I will update it in changelog. So I close this issue now

closed 2021-03-12 by zlj-zz