elephantrobotics/pymycobot #11: pwm cannot set Hz

shiosalt, 2021-03-01:
**Issue
pymycobot' PWM function cannot set Hz.
So I cannot use my servo for my self-made grip.
My servo needs 50Hz.
By my analysis, PWM 's frequency is fixed in about 1.8KHz.

**To Reproduce**
run python script below
-------------------
from pymycobot.mycobot import MyCobot
GPIO_Pin_No=23
PWM_channel_No=0
PWM_Duty_Cycle=128 # 
[image]
50%
mc = MyCobot('/dev/ttyUSB0')
mc.set_pwm_mode( GPIO_Pin_No, PWM_channel_No)
mc.set_pwm_output(PWM_channel_No , PWM_Duty_Cycle)
--------------------
**Expected behavior**
for example,
-------------------
from pymycobot.mycobot import MyCobot
GPIO_Pin_No=23
PWM_channel_No=0 
PWM_Duty_Cycle=128 # 50%
PWM_Freq=50 # Hz
mc = MyCobot('/dev/ttyUSB0')
mc.set_pwm_mode( GPIO_Pin_No, PWM_channel_No)
mc.set_pwm_output( PWM_channel_No , PWM_Duty_Cycle , PWM_Freq)
--------------------

zlj-zz, 2021-03-06:
Atom2.6 should be ok.

closed 2021-03-06 by zlj-zz

shiosalt, 2021-03-09:
I upgraded Atom to 2.6, but it doesn't work.
 I think it's a Python issue, not an atom issue. 

python command's output:
`>>> mc.set_pwm_output( PWM_channel_No , PWM_Duty_Cycle , PWM_Freq)`
`Traceback (most recent call last):`
 ` File "<stdin>", line 1, in <module>`
`TypeError: set_pwm_output() takes 3 positional arguments but 4 were given`

shiosalt, 2021-03-11:
Unfortunately, Atom 2.6 cannot work with 2 args. I tested 2 times.

I guess Atom 2.6 needs 3 args, but pymycobot sends only 2 args.

 *Transponder app send "FE FE 04 64 (PWM_channel_No) (PWM_Duty_Cycle) FA"
   on  set_pwm_output(PWM_channel_No , PWM_Duty_Cycle),
  and Atom 2.6 doesn't make waves.  Atom 2.5 makes waves.

zlj-zz, 2021-03-12:
OK, I see. Atom 2.6 does need three parameters, but it's not open to me yet, so I can't improve pymycobot. 
When atom solves this problem, I will also update pymycobot。