elephantrobotics/pymycobot #13: Real-Time Synchronization of movement

tkdsym2, 2021-05-21:
**Describe the bug**
myCobot reads the slider values and does not run smoothly.

**To Reproduce**
I set up an OSC server in python and send values using the pymycobot module.
Here is a part of the function. However, I just use prepared functions.

```
def sync_angle(self, _angles, _speed):
        self.arm.sync_send_angles(_angles, _speed)
```

Here is a video recording of the operation of the slider and the actual movement.
At 1 minute and 7 seconds, I try to move the slider while holding it, but mycobot does not move smoothly.
https://www.youtube.com/watch?v=13HlRCM19ws

**Expected behavior**
It is expected that the position and angle of myCobot will move smoothly according to the position of the slider.
It is also hoped that a tracking device such as Leap Motion can be used to synchronize the position of the mycobot with the position of one's hand by continuously sending values in real time.

**Screenshots**
Recording Video:
https://www.youtube.com/watch?v=13HlRCM19ws

**Desktop (please complete the following information):**
 - OS: Windows 10 Home
 - Python 3.8.5
 - pymyCobot 2.4.1
 - python-osc 1.7.4
 - OSC signal from Touch Designer 2021.12380

**Additional context**
myCobot is Tranponder mode.

zlj-zz, 2021-05-24:
This `sync` method relies on `is_in_position()`, but `is_in_positon` is not so stable.
Serial communication contains packet loss and delay, so this may be the reason why mycobot can not run smoothly.

tkdsym2, 2021-05-24:
@zlj-zz 
Thank you very much for your feedback.
I am using [the official Transponde](https://github.com/elephantrobotics/myCobot/blob/main/Arduino/MycobotBasic/examples/Transponder/Transponder.ino)r program. The rate is 115200, it means fastest. So you mean that in order to achieve smoother movement, the Transponder program needs to be modified or the signals sent by the python program need to be better controlled.

In addition,  I just saw a couple of samples that realize the smooth movement ([1](https://twitter.com/OTL/status/1351345117697695744?s=20), [2](https://twitter.com/Wu_Shengzhi/status/1396120848126210049?s=20)). One of them uses ROS, while the other one looks like it implements the original program.
Do they cover packets and delays, and the losses therein?
I have checked [the script to control myCobot from ROS](https://github.com/elephantrobotics/mycobot_ros/blob/main/scripts/slider_control.py). However, it looks like it just `sends_radians` [here](https://github.com/elephantrobotics/mycobot_ros/blob/main/scripts/slider_control.py#L22). Is there something else you are doing on the ROS side?

Do you have any ideas on how to send signals from python to myCobot to make it run smoothly?

zlj-zz, 2021-05-25:
@tkdsym2 
The probability of packet loss is very small.The question is on `is_in_position()`.
According to my current test, this method cannot return to `true` immediately after the manipulator reaches the point.

The manipulator may not be able to reach the target point completely due to the large force distance, and the error between the real angle and the target angle is too large (for example > 5mm), so `is_in_position()` can not judge the position successfully. So you may need to wait until the timeout (the default is 7S).

If you want to achieve a function similar to tracking, then I believe your data should be continuous and the time interval is very short.So you may not need to use this method. You can try to send data continuously (e.g. angles), as follows:
```python
while True:
    angles = queue.get() # Getting data from a queue.
    mycobot.send_angles(angles, 90)
    time.sleep(.05)
```
It should also be noted that the time taken to send the data itself is about 5ms(e.g. `send_angles()` or something).

tkdsym2, 2021-06-02:
@zlj-zz 
Thank you very much for your great feedback, and I am sorry to be late.

I had been trying to use your example. However, the movement of myCobot was the same.
I just pushed [my codes ](https://github.com/tkdsym2/myCobot-realtime-movement)using Processing instead of TouchDesigner. It is easier and more simple than using TouchDesigner. The code just sends one angle based on the parameter from the slider on Processing program.

Could you please see if what I am doing is the same as what you have shown me? Also, it would be great if you could actually run the code to see what is happening.

thanks!

zlj-zz, 2021-06-02:
I am sorry, I don't have a Processing environment.
But I looked at your code and it didn't look like a problem.

But the following code will read the serial port 9 times, too much fast reading may lead to empty data return. It seems that once is enough. 
```python
    while True:
                if connection:
                    simpleClient.send_message('/util/status/angles', arm.get_arm_status()['angles'])
                    simpleClient.send_message('/util/status/coords', arm.get_arm_status()['coords'])
                    simpleClient.send_message('/util/status/radians', arm.get_arm_status()['radians'])
                    # print('{}'.format(arm.get_arm_status()))

                asyncio.sleep(0.01)
```

BTW. Do you have any other questions? This doesn't seem to be tracking.

zlj-zz, 2021-07-07:
If there is no activity for a long time, this issue will be closed.

closed 2021-07-07 by zlj-zz