elephantrobotics/pymycobot #18: Gripper does not keep closed

junjungoal, 2021-08-01:
**Describe the bug**
The gripper does not keep closed (a servo motor does not keep running), and, it drops an object in the gripper.

**To Reproduce**
Steps to reproduce the behavior:
1. Burn firmware using MyStudio
 - On Basic board: `minirobot` v0.4 
 - On Atom board: `AtomMain` v2.8  (I also tried latest version and older version currently available, but none of these worked)
2. Install this package
3. Use `Transponder`
4. Calibrate the gripper by calling `set_gripper_ini()` after I fully open the gripper. 
5. Execute the following script
```
from pymycobot.mycobot import MyCobot
mc = MyCobot('/dev/ttyUSB0')
mc.set_gripper_state(1, 70)
```
4. A gripper keep closed for some seconds (sometimes around 30~40s, or even around 10s), but eventually it repeats loosing and closing (you can check how it behaves in the attached movie ).. which results in releasing an object that the gripper grasps. 

**Expected behavior**
The gripper should keep closed all the time. That is, a servo motor should keep running to grasp an object.....otherwise it releases the object.

**Screenshots**

https://user-images.githubusercontent.com/9709386/127781900-bff2efdb-c8b8-414c-9a57-ff5c8a1c83be.mp4

**Desktop (please complete the following information):**
 - Ubuntu
 - 18.04

zlj-zz, 2021-08-04:
This is because the gripper is too tight.

There are several ways you can try:
- Update your Atom to 3.*
- Refer to [https://github.com/elephantrobotics/pymycobot/blob/main/demo/gripper.py](https://github.com/elephantrobotics/pymycobot/blob/main/demo/gripper.py) and use `set_encoder`

junjungoal, 2021-08-04:
@zlj-zz 
I see. 
So basically, I may need to change the second argument of `set_encoder` function to make sure that the gripper won't be too tight... as it did not work with 1200, but worked with 1500 in my case. 

Thank you very much.

closed 2021-08-04 by junjungoal