elephantrobotics/pymycobot #19: set_encoders() executes get_encoders() functionality and get_encoders() returns empty array

3110, 2021-08-04:
**Describe the bug**
* `set_encoders()` definition is duplicated, so `set_encoders()` executes `get_encoders()` functionality.
* `get_encoder()` returns empty array.

**To Reproduce**
```
In [1]: from pymycobot.mycobot import MyCobot
In [2]: mc = MyCobot("COM6")
In [3]: mc.set_encoders()
Out[3]: []

In [4]: for i in range(1, 7):
   ...:     print(i, mc.get_encoder(i))
   ...:
1 [913]
2 [415]
3 [416]
4 [2776]
5 [2667]
6 [2078]
```

**Expected behavior**
```
In [1]: from pymycobot.mycobot import MyCobot
In [2]: mc = MyCobot("COM6")
In [3]: mc.get_encoders()
Out[3]: [913, 415, 416, 2776, 2667, 2078]
```
**Desktop:**
 - OS: Windows 11
 - pymycobot: 2.5.6
 - AtomMain: 3.2

zlj-zz, 2021-08-05:
Thank you report. I have fixed it. Please try v2.5.7

3110, 2021-08-05:
This bug has not been fixed in v2.5.7. I think you have to implement some codes to parse the reply from the command GET_ENCODERS in MyCobot::_mesg.

zlj-zz, 2021-08-05:
Is ok `get_encoder`?

3110, 2021-08-05:
Yes, the `get_encoder()` bug (#20) has been fixed. Thanks.

zlj-zz, 2021-08-06:
I test the `get_encoders()`, seem is right?

```
─4─[~/P/e/pymycobot]─[main*]── ─ pytest -s tests/test_api.py::test_encoder                                                                 -- INS --
================================================================ test session starts ================================================================
platform darwin -- Python 3.9.4, pytest-6.2.3, py-1.10.0, pluggy-0.13.1
rootdir: /Users/zacharyzhang/Projects/elephantrobotics/pymycobot
collected 1 item                                                                                                                                    

tests/test_api.py 
1 : /dev/cu.wlan-debug - n/a
2 : /dev/cu.debug-console - n/a
3 : /dev/cu.Bluetooth-Incoming-Port - n/a
4 : /dev/cu.usbserial-0243F266 - CP2104 USB to UART Bridge Controller

Please input 1 - 4 to choice:4
/dev/cu.usbserial-0243F266

Please input baudrate (default: 115200):
115200

Wether DEBUG mode[Y/n] (default: no):n

[2075, 3640, 3806, 2334, 977, 639]
```

3110, 2021-08-06:
Umm, in my environment with v2.5.7, `get_encoders()` has still returned an empty array. It seems that you test on your local pymycobot , so do you commit your fix for this bug?

```
$ ipython
Python 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.20.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from pymycobot.mycobot import MyCobot

In [2]: mc = MyCobot("COM6")

In [3]: mc.get_encoders()
Out[3]: []  # expected [1174, 3668, 3770, 3747, 2687, 1561]

In [4]: for i in range(1, 8):
   ...:     print(i, mc.get_encoder(i))
   ...:
1 1174
2 3668
3 3770
4 3747
5 2687
6 1561
7 2069  # gripper
```

zlj-zz, 2021-08-06:
I confirm that this amendment was submitted long ago.
Can you try to install the source code? Is there a problem with the released package.

3110, 2021-08-06:
Though I try to install from the source code, the result is the same.   I investigate more detail, and I find that AtomMain v3.2 does not reply to the GET_ENCODERS command.  What version of AtomMain do you use?

zlj-zz, 2021-08-06:
Thank you for waiting. According to your question, we tested again, and the result is no problem.

-  AtomMain version: 3.2

Didn't you try to read more times? Or re-flash the Atom？

zlj-zz, 2021-08-06:
It's really atom's problem. Did you try 3.1?

3110, 2021-08-06:
Though I try to v3.1,  and all versions available on myStudio, the results are the same...

zlj-zz, 2021-08-06:
Thank you for your feedback. We have now identified it as an atom problem. I use the version under development. It's OK. It's a problem with our release version.

fedorenchik, 2021-09-18:
Fixed in pymycobot 2.5.9.

closed 2021-09-18 by fedorenchik