elephantrobotics/pymycobot #21: Unnecessary fix for is_servo_enable() in v2.5.8

3110, 2021-08-10:
**Describe the bug**
The fix 4b4a14419463067e6f788c77598688118bce9b66 is unneccesary. You have to fix the reply of is_servo_enable() from AtomMain v3.2 instead.

**To Reproduce**
AtomMain v3.2 replies the unnecessary joint number(6) for `is_servo_enable(6).` 
```
Sending:  fe fe 03 50 06 fa
Received: fe fe 04 50 06 01 fa
```

**Expected behavior**
```
Sending:  fe fe 03 50 06 fa
Received: fe fe 03 50 01 fa
```
Other commands that specified a joint number do not include it in their reply.

**Desktop:**
- myCobot-280(M5stack ver.)  
  - AtomMain v3.2
  - minirobot v0.4
- OS: Windows 11  
  - Python v3.9.6
  - pymycobot v2.5.8

zlj-zz, 2021-08-12:
I'm not sure why, but that's what we designed.

zlj-zz, 2021-08-12:
I asked my colleagues. This is used to confirm information and prevent loss during asynchronous communication.

3110, 2021-08-12:
Thanks, I understand what you intend to.

closed 2021-08-12 by 3110