elephantrobotics/pymycobot #23: jog_coord not work

MasashiOhtake, 2021-08-31:
**Describe the bug**
jog_coord() command not work in myCobot Pi

**To Reproduce**
Steps to reproduce the behavior:
from pymycobot.mycobot import MyCobot
from pymycobot.genre import Coord
from pymycobot.genre import Angle
from pymycobot import PI_PORT, PI_BAUD  # When using the Raspberry Pi version of myCobot, you can refer to these two variables for MyCobot initialization
import time
# Initialize a mycobot object
mc = MyCobot(PI_PORT, PI_BAUD)
mc.power_on()
mc.jog_coord(1,1,50)
time.sleep(1.5)

**Expected behavior**
my cobot move

**Desktop (please complete the following information):**
 - myCobotPi
 - OS: Raspbian GNU/LINUX 10 (buster)
        pymycobot 2.5.9
        python 2.7.16

**Additional context**
Add any other context about the problem here.
Using jog_angle command can move mycobot

Eheech, 2021-08-31:
Sorry, we haven't release this function yet ... We suggest you use send_coord() or send_coords() first ...

MasashiOhtake, 2021-08-31:
Thanks for your kind comment

closed 2021-08-31 by MasashiOhtake