elephantrobotics/pymycobot #28: pymycobot::get_angles may fail depending on current angle condition

SidKoe, 2021-11-04:
**Describe the bug**
get_angles() may fail.

**To Reproduce**
changes angles using pymycobot API.
and then call get_angles().
It will empty array as result.

**Desktop (please complete the following information):**
Python 3.8.7

**Additional context**
I looked into this issue and found strange behavior as following.
When host acquires angles, host receives response from MyCobot contains servo's response and M5atom response.
Servo's response terms with check sum and it is 0xfe.
Response from M5stom starts with 0xfe, 0xfe two bytes.
At this time. DataProcessor::_process_received() makes mistake on finding header of response of M5atom.
I quick fixed by taking a look data in after header if it is same as 'genre'. If it is not same, it continues from next byte sequence.

Eheech, 2021-11-05:
Would you mind uploading your codes ? We tried send_angles() and get_angles() multiple times , didn't have this issue , so if you don't mind, we'd like to test it with your code , thanks ~

SidKoe, 2021-11-05:
I am used to using GitHub so I attach changed code here.
Code I have changed is as below. It just DataProcessor::_process_received() only.
Changed lines is marked with command, "# CHANGED"
I also attach binary dump of transmitting data and comment following code. I hope it helps you to understand issue.

---- CODE:start ----
    def _process_received(self, data, genre):
        if not data:
            return []

        data = bytearray(data)
        data_len = len(data)
        # Get valid header: 0xfe0xfe
        for idx in range(data_len - 1):
            if self._is_frame_header(data, idx):
                data_len = data[idx + 2] - 2
                if data_len > 0:
                    cmd_id = data[idx + 3]   # CHANGED
                    if cmd_id == genre:       # CHANGED
                        break                         # CHANGED
        else:
            return []

        data_len = data[idx + 2] - 2

        # compare send header and received header
        cmd_id = data[idx + 3]
        if cmd_id != genre:
            return []
        data_pos = idx + 4
        valid_data = data[data_pos : data_pos + data_len]

        # process valid data
        res = []
        if data_len == 12:
            for idx in range(0, len(valid_data), 2):
                one = valid_data[idx : idx + 2]
                res.append(self._decode_int16(one))
        elif data_len == 2:
            res.append(self._decode_int16(valid_data))
        else:
            res.append(self._decode_int8(valid_data))
        return res
---- CODE:end ----

Each line is divided by chunk of command or response.
Data start with servo related communication data and last chunk is response from M5atom, I thought.
Last servo response terms with check sum and it value is 0xfe.
Succeeding M5atom response start with 0xfe, 0xfe, header part.
Current code will take termination check sum code and first one byte of data as starting response of M5atom.
And pymycobot library will check command code at wrong data and fail parsing.
---- DATA:start ----
ff ff 01 04 02 38 02 be 
ff ff 01 04 00 87 06 6d 
ff ff 02 04 02 38 02 bd 
ff ff 02 04 00 f1 03 05 
ff ff 03 04 02 38 02 bc 
ff ff 03 04 00 43 07 ad 
ff ff 04 04 02 38 02 bb 
ff ff 04 04 00 70 03 84 
ff ff 05 04 02 38 02 ba 
ff ff 05 04 00 11 08 dd 
ff ff 06 04 02 38 02 b9 
ff ff 06 04 00 f0 07 fe 
fe fe 0e 20 0c f1 23 ab f9 8c 28 19 ff 6B 00 8c fa
---- DATA:end ----

SidKoe, 2021-11-05:
P.S.
Well, I think you may repro more easier when you change angle of J6. I think last response is for J6 servo. Others may not cause this issue. If you have any question, please let me know.

Eheech, 2021-11-05:
In pymycobot , it will filler out communication messages between Basic and Atom , only read Get_angles()
[image]

Eheech, 2021-11-05:
When you get empty array, have you check Atom's status ? Is it 'yes' or 'no' ? and have you chosen  ' minirobot -> transponder ' ?

Eheech, 2021-11-05:
Maybe you can upload your code which you used to control mycobot , we'd like to test it with your code , thanks ~

SidKoe, 2021-11-05:
I have not yet checked Atom's status. I am not sure which API is available to see Atom's status.  But even if this issue happened mycobot can be controlled from pymycobot API; changing angle of joints, focus on or off servos. I believe this issue is not problem of Atom or Base. It will be handling problem of received data in pymycobot or something. 
I am using transponder and set various angles of all joints. Most cases it works fine; get_angles() returns list of angles of joints.
BTW, it is difficult to provide code controlling mycobot as it is, because it includes many functions for another project. I will try to arrange something to provide for this issue.

Eheech, 2021-11-05:
Are you using mycobot_280_M5 ? In transbonder, press button A , it will return Atom's status in Basic.

tongtybj, 2022-01-16:
Hi, I face the same with MyCobot Pro 320 M5.
My code is based on mycobot_ros.
The debug branch is https://github.com/tongtybj/mycobot_ros/tree/interface_bug_mycobot320pro_m5.

v-ghobadi, 2023-10-26:
myCobot 280pi returns empty array for get_angles().

michelle-aa-robotics-photonics, 2024-07-02:
I have also had this issue of receiving an empty array when calling `get_angles()`. I previous did not have this issue, but now I do and I'm not sure why.

closed 2024-11-07 by anla-xu