elephantrobotics/pymycobot #39: get_angles return empty

Kakcalu13, 2022-10-24:
**Describe the bug**
Similar to https://github.com/elephantrobotics/pymycobot/issues/19, but this worked for the first time but then it stopped return data after that. Restart mycobot doesn't fix the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '~/pymycobot/demo'
2. python3 basic.py
3. Enter all data like input of usb, baud and debug mode. After that, `get_angles()` should return data, not empty.
4. See error
```

rpi@raspberrypi:~/pymycobot/demo $ python3 basic.py 

--------------------------------------------
| This file will test basic option method: |
|     set_led_color()                      |
|     send_angles()                        |
|     get_angles()                         |
|     send_angle()                         |
|     send_radians()                       |
|     get_radians()                        |
|     send_coords()                        |
|     get_coords()                         |
|     send_coord()                         |
--------------------------------------------
          

1 : /dev/ttyUSB0 - CP2104 USB to UART Bridge Controller - CP2104 USB to UART Bridge Controller
2 : /dev/ttyAMA0 - ttyAMA0

Please input 1 - 2 to choice:1
/dev/ttyUSB0

Please input baud(default:115200):115200
115200

Wether DEBUG mode[Y/n]:n

Start check basic options

::set_color() ==> color 255 255 0

::send_angles() ==> angles [0, 0, 0, 0, 0, 0], speed 100

::get_angles() ==> degrees: []

::send_angle() ==> angle: joint1, degree: 90, speed: 50

::send_radians() ==> set radians [1, 1, 1, 1, 1, 1], speed 100

::get_radians() ==> radians: []

::send_coords() ==> send coords [160, 160, 160, 0, 0, 0], speed 70, mode 0

::get_coords() ==> coords []

::send_coord() ==> send coord id: X, coord value: -40, speed: 70

::set_free_mode()

=== check end ===

```

**Expected behavior**
It should return the value of degree data

wildangunawan, 2023-04-03:
Same issue using 280 Pi

wildangunawan, 2023-04-03:
I already send this concern to the tech dept (support at elephantrobotics dot com) email. Hopefully we get response soon.

anla-xu, 2023-04-20:
1.What is the probability that this problem occurs? Occasionally or all the time？

wildangunawan, 2023-04-20:
> 1.What is the probability that this problem occurs? Occasionally or all the time？ 2.Please provide the model of the machine with the problem and the software version used (like atom's firmware version)

1. All the time.
2. myCobot 280 Pi 2020, need to check the software version but I think would be the latest one.

anla-xu, 2023-04-20:
> > 1.出现这个问题的概率是多少？偶尔还是一直？ 2.请提供问题机器的型号和使用的软件版本（如atom的固件版本）
> 
> 1. 每时每刻。
> 2. myCobot 280 Pi 2020，需要检查软件版本，但我认为应该是最新的。

Can other interfaces be used? For example get_coords, is_power_on, The baud rate of 280 Pi is 1000000. Is this how you initialized it?
```
from pymycobot import MyCobot

mc = MyCobot('/dev/ttyAMA0', 1000000)
```

wildangunawan, 2023-04-20:
> Can other interfaces be used? For example get_coords, is_power_on, The baud rate of 280 Pi is 1000000. Is this how you initialized it?
>
> ```
> from pymycobot import MyCobot
>
> mc = [...]

Yes, others can be used but get_angles cannot. Considering we're playing with angles here, I think this is an important feature to have.

I initialized the myCobot using [this](https://github.com/elephantrobotics/pymycobot#source-code) `from pymycobot import PI_PORT, PI_BAUD # For raspberry pi version of mycobot.`

anla-xu, 2023-04-20:
> > > > 1.出现这个问题的概率是多少？偶尔还是一直？ 2.请提供问题机器的型号和使用的软件版本（如atom的固件版本）
> > > 
> > > 
> > > 
> > > 1. 每时每刻。
> > > 2. myCobot 280 Pi 2020，需要检查软件版本，但我认为应该是最新的。
> > 
> > 
> > Can other interfaces be used? For example get_coords, is_power_on, The baud rate of 280 Pi is 1000000. Is this how you initialized it?
> > ```
> > from pymycobot import MyCobot
> > 
> > mc = MyCobot('/dev/ttyAMA0', 1000000)
> > ```
> 
> Yes, others can be used but get_angles cannot. Considering we're playing with angles here, I think this is an important feature to have.
> 
> I initialized the myCobot using [this](https://github.com/elephantrobotics/pymycobot#source-code)
> 
> ```
> from pymycobot import MyCobot
> from pymycobot import PI_PORT, PI_BAUD # For raspberry pi version of mycobot.
> mc = MyCobot(PI_PORT, PI_BAUD)
> ```

Please update pymycobot to the latest version, and add `debug=True` parameters during initialization to see if there is any data returned, please send me the running results
```
from pymycobot import MyCobot
from pymycobot import PI_PORT, PI_BAUD # For raspberry pi version of mycobot.
mc = MyCobot(PI_PORT, PI_BAUD, debug=True)
print(mc.get_angles())
```

v-ghobadi, 2023-10-26:
Why get_angles() don't work for myCobot 280pi?

anla-xu, 2024-04-22:
@Kakcalu13 Please provide the basic firmware, atom firmware and pymycobot version number you are using.

michelle-aa-robotics-photonics, 2024-07-01:
I am also having this issue. I first was able to use the bot and extract the angles. Now I am not. The only change is that I updated myStudio and down
[image]
[image]
[image]
[image]
[image]
loaded the latest Atom firmware. I did not flash that firmware, though, because if I press the "Flash" button I get "Please connect the machine before burning". I checked the firmware and I have version 5.1.

closed 2024-11-07 by anla-xu