elephantrobotics/pymycobot #40: Fresh Ubuntu 20 installation in RPi can't communicate with Atom

saratrajput, 2022-10-29:
**Describe the bug**
I have installed Ubuntu 20 on a new SD card for the RPi of mycobotpi. I have installed pymycobot package with pip as well. But I'm not able to communicate with the robot and can't execute any motion.

I'm wondering what are the steps required when making a fresh Ubuntu installation on the RPi. Are there any packages that need to be installed? Where can I find more documentation regarding this.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]

 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

saratrajput, 2022-11-15:
@anla-xu Thank you for your reply. It is the official system I downloaded myself. Is there a system provided by you as well? I wasn't aware of that. The OS on my original RPi was Raspbian Buster.

anla-xu, 2022-11-15:
@saratrajput You can't directly use the official system of Raspberry Pi to control the mechanical arm. You need to reconfigure the serial port. You can try to search on Google: how to configure the serial port of Raspberry Pi, which may be difficult. The most effective way is to find after-sales customer service

closed 2022-11-22 by anla-xu