elephantrobotics/pymycobot #61: Specific command does not work on M1 Mac unlike Linux Desktop/Raspberry PI

Kakcalu13, 2024-05-10:
I made a reproducible the bug. See here:
```
from pymycobot.mycobot import MyCobot
import time

class Arm:
    @staticmethod
    def connection_initialize(port='/dev/cu.usbserial-023EDC85'):
        """
        :param port: The default would be '/dev/ttyUSB0'. If the port is different, put a different port.
        :return:
        """
        return MyCobot(port, 115200)

    @staticmethod
    def pose_to_default(arm, count):
        for number_id in range(1, count, 1):
            if number_id != 2: # my #2 servo is broken xP
                arm.set_encoder(number_id, 2048)

# DEBUG SECTION
mycobot = Arm() # INITALIZED THE ARM
arm = mycobot.connection_initialize() # default to /dev/ttyUSB0 on linux/RPI. If mac, use different port.
print("initalized the arm")
time.sleep(2)
arm.set_speed(100) # Just to make it move fast.
print("moving arms to default position.")
time.sleep(2)
mycobot.pose_to_default(arm, 7) # 1 to 6 servo. This is an issue for Mac M1 on latest python: 3.0.6
# Linux is fine on this. It's an issue for Mac only
print("about to release all servos at once")
time.sleep(3)
arm.release_all_servos()
```

I added more simple code. The issue is that `set_encoder` does not work well on Mac M1 for some reason. I ran the exact same code above on linux except the different port name which is `/dev/USB0` and it worked. 

Mac and Linux used the same version of mycobot.

On Mac information:
Pymycobot Version: 3.0.6
Mac CPU: Apple M1
Mac OS: Version 13.6.4 (22G513)
Python version: Python 3.11.3

On my Linux:
Pymycobot Version: 3.0.6
Linux CPU: Intel i7 CPU
Linux OS: 20.04
Python version: Python 3.8.10

My mycobot arm version is mycobot pi 280

anla-xu, 2024-05-11:
If you want to use a PC to control 280 Pi, you need to use Socket communication. PC cannot directly control 280Pi through USB. 

You need to execute the [Server.py](https://github.com/elephantrobotics/pymycobot/blob/main/demo/Server.py) file in a 280Pi system, open the server, and then use [MyCobotSocket](https://github.com/elephantrobotics/pymycobot/blob/main/docs/README.md#mycobotsocket) on the PC to connect to the client and control the robot

Kakcalu13, 2024-05-14:
Sorry for the delay!

I'm confused, I've been controlling the mycobot through USB on PC for almost a year now. It's just that MAC doesn't work. So Mac needs to be the one to connect with the server? How come it worked on Linux, but not Mac?

anla-xu, 2024-05-15:
I'm curious, how did you implement the PC to communicate with the 280 Pi if you didn't use the Socket method? Only the M5 series mycobot can use USB to communicate with PC. The type-c port on the 280 Pi is not used for communication. If you want to use a MAC computer to control the 280 Pi, you can only use Socket.

closed 2024-06-06 by anla-xu