elephantrobotics/pymycobot #66: mc.get_coords() returns None

y-z-g, 2024-06-07:
**Describe the bug**
Following the myCobot 280 M5 demo code, got error.

**To Reproduce**
Steps to reproduce the behavior:
1. pip install pymycobot --upgrade
2. copy paste demo/myCobot_280_demo/280_draw_gcode.py and modify the usb port for linux use
3. run demo script

**Expected behavior**
demo code should work but mc.get_coords() returns None.

**Env (please complete the following information):**
ubuntu, python3.11 venv

lalado-o, 2024-06-12:
Did you select **Transponder > USB UART > OK** before running the demo script?
The robot needs to establish a connection before it can be programmed

y-z-g, 2024-06-12:
Select menu:
miniRoboFlow v2.30 -> Transponder -> USB UART

Showing:
Connect test
Atom: OK

anla-xu, 2024-06-19:
You can write a simple test script like this to see if there is any output。
```python
from pymycobot import MyCobot
import time
m = MyCobot("com5")

for _ in range(10):
    print(m.get_angles())
    time.sleep(0.1)
    print(m.get_coords())
    time.sleep(0.1)
```

closed 2024-07-22 by anla-xu