elephantrobotics/pymycobot #72: send_angles does not behave as I thought it would

michelle-aa-robotics-photonics, 2024-07-02:
**Describe the bug**
I am starting at a joint position of `[-12.21, -70.48, -71.19, -32.51, -104.94, 3.6]` for a 6-DOF cobot Raspberry Pi. I then send `send_angles([[-7.82, -85.69, -26.63, -65.47, -95.36, 3.6]`), in the understanding that it will decrease or increase the corresponding joint to achieve this new position. However, when I do this, joint 5 increases instead of decreases and the bot ends up getting stuck at -105 degrees. Am I understanding something wrong or is this buggy behavior that needs to be fixed?

**To Reproduce**
Steps to reproduce the behavior:
1. Move bot to `[-12.21, -70.48, -71.19, -32.51, -104.94, 3.6]`
2. Send `send_angles([[-7.82, -85.69, -26.63, -65.47, -95.36, 3.6]`
3. See bot increase joint 5

**Expected behavior**
All joints should increase or decrease based on the numerical diff between current and target position.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspberry Pi Ubuntu 18.04
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

anla-xu, 2024-07-03:
From the two sets of data you sent, J5 of the first set is -104.94, and J5 of the second set is -95.36. It is normal for the value to increase from -104.94 to -95.36. I don't understand your question. Please provide more detailed information.

michelle-aa-robotics-photonics, 2024-07-03:
Hi, I'm sorry! That was two typos -- J5 decreased, and it got stuck at -150 (not 105).

anla-xu, 2024-07-03:
After sending this set of data [-7.82, -85.69, -26.63, -65.47, -95.36, 3.6], does the robot move? And after the movement is over, please use `get_angles` to read the data to see what it is

michelle-aa-robotics-photonics, 2024-07-03:
After turning the bot off and on again, the bot does move! It curls up into itself and it is stuck like the below picture. I used `get_angles()` and it returned `[-7.47, -44.38, -16.34, -65.83, 45.87, 4.48]`
[image]
I think this is because I have re-calibrated a new bot since submitting this error and it probably has very different joint angles from default settings.

michelle-aa-robotics-photonics, 2024-07-03:
[image]
I just tried to re-calibrate to set new zero position, but the bot did not move at all during the calibration routine.

michelle-aa-robotics-photonics, 2024-07-03:
I finally fixed this by closing the test tool and re-opening b/c I had accidentally set the connection type to M5 instead of Raspberry Pi. After resetting the angles, I managed to send one successful `send_angles()` command:
```
mc.get_angles() = [111.88, -68.37, -39.55, -63.45, -158.11]
mc.send_angles([115.57, -27.68, -112.32, -47.98, -147.04, 3.51], 1)
```
produced this video!!!

https://github.com/elephantrobotics/pymycobot/assets/173008148/5f73890d-6a06-41c0-a3e2-db5705f2508e

However, after that I no longer got any response after trying to send:
```
mc.send_angles([115.57, -27.68, -112.32, -45, -147.04, 3.51], 1)
mc.send_angles([115.57, -27.68, -112.32, -41, -147.04, 3.51], 1)
```
This is the problem I keep running into. The bot responds to this command once and then never again. What am I doing wrong?

anla-xu, 2024-07-05:
Please set the angle to zero send_angles([0,0,0,0,0,0], 20). After the movement is over, it is normally in a straight state, and the depressions at the joints are close to alignment. Please provide me with a picture.

closed 2024-08-13 by anla-xu