elephantrobotics/pymycobot #73: Out-of-the-box cobot 280 Pi does not move bot when I command send_angles

michelle-aa-robotics-photonics, 2024-07-02:
**Describe the bug**
Got a new cobot 280 Pi, set it up and hooked it up to my monitor. The arm seems to be able to read angles through `get_angles()` but it doesn't `send_angles()` successfully. As in, it thinks it has but the robot doesn't move. And if you `get_angles(`) after you `send_angles()` , then it thinks it's updated the angles and returns the new coordinates. But it's wrong because the robot has not moved. If you release the servos and drive the motor around again, it seems to reset itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Move joint angles to `[79.45, -50.44,  -92.02, -33.13, 69.54, -41.92]`
2.  Send `send_angles([76.2, -43.41, -114.96, -21.26, 68.22, -42.27])`
2. See error that bot does not move
3. Send `get_angles()`
5. See error that bot has updated the joint angles it returns even though it has not moved.

**Expected behavior**
If I send `send_angles([76.2, -43.41, -114.96, -21.26, 68.22, -42.27])`, I expect the bot to move to those coordinates.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspberry Pi 18.04
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

anla-xu, 2024-07-03:
You can use the `send_angle` interface to verify that each joint can move normally. If you want to control the coordinates, please use the `send_coords` interface.

michelle-aa-robotics-photonics, 2024-07-03:
Hi! I did use the `send_angle` interface and it also didn't move the joints.

I misspoke early when I said "coordinates". All I've been checking are the joints.

anla-xu, 2024-07-03:
Please use myStudio to update the atom firmware and try again. Burning tutorial:https://docs.elephantrobotics.com/docs/mycobot-pi-en/4-BasicApplication/4.1-myStudio/4.1.1-myStudio_download_driverinstalled.html

michelle-aa-robotics-photonics, 2024-07-03:
I'm a bit confused -- this page does not seem to have any tutorial on updating Atom firmware, it is about building myStudio?

If you are referring to https://docs.elephantrobotics.com/docs/mycobot-pi-en/4-BasicApplication/4.1-myStudio/4.1.2-myStudio_flash_firmwares.html, these instructions all seem to refer to plugging directly into a computer, which I don't believe the Cobot 280 Pi can do.

anla-xu, 2024-07-04:
You need a USB to type-c converter and buy a communication cable. Connect the type-c end to the atom port at the end of the robot arm. The USB end can be connected to your computer or 280 Pi, then start myStudio to burn the firmware.

closed 2024-08-13 by anla-xu