elephantrobotics/pymycobot #79: Mercury B1 - Joints 11-13 Stuck at Full Timeout

shale-bridge, 2024-07-31:
Moving joints 1-7 using send_angle moves the joint and blocks until the move is complete. Moving joints 11-13 will move the joint but block in a timeout loop (300 seconds).

**To Reproduce**
```
# initialize the arm instances
armLeft = Mercury("/dev/left_arm")
armRight = Mercury("/dev/right_arm")

# move an arm joint +20 degrees
armRight.send_angle(1, 20, 40) # this works fine, and returns once move is complete

# try to move the neck joint to +20 degrees
armRight.send_angle(12, 20, 40) # this blocks for a full timeout value of 300 seconds
```

**Expected behavior**
Control should return to the program when move is complete, like joints 1-7.

**Hardware Information**
Running on Mercury B1 Ubuntu using the Python API.

pymycobot version is "3.5.0b8" as listed in the __init__.py file that was loaded onboard. Curiously, the files look very different to the repo here on github. While debugging, the problem is in file "mercury_api.py", in function "_mesg". Wait_time gets set to 300, and only on joints 11-13 does it time out stuck in a loop. 

The files on here are quite different (and latest release is "3.4.6"!) and there is no wait time set to 300 anywhere.

anla-xu, 2024-08-01:
I'm very sorry, but Mercury's Python library is still in the testing phase and is not complete. It is different from the main branch. We will inform you of the update after the subsequent testing is completed.

anla-xu, 2024-08-16:
You can update pymycobot to `v3.5.0b17` to avoid the problem of long-term stuck during control. 

```shell
pip install pymycobot==3.5.0b17
```

> Note: This version needs to increase the delay after sending the motion command.

For example：
```python
from pymycobot import Mercury
import time

ml = Mercury("/dev/left_arm")
ml.send_angle(1, 0, 20)
time.sleep(3)
ml.send_angle(1, 90, 20)
time.sleep(3)

```

afishtex, 2024-09-05:
> You can update pymycobot to `v3.5.0b17` to avoid the problem of long-term stuck during control.
> 
> ```shell
> pip install pymycobot==3.5.0b17
> ```

Hello, I attempted the suggested update on the same system that @shale-bridge is using and now the robot's right arm, which was perfectly functional, no longer responds to commands. When using: 

```
mr = Mercury("/dev/right_arm", debug` = True)
mr.power_on()
```

I can see that the code is writing commands to the port, but not reading them back. 

`17:08:59.816 DEBU [pymycobot.generate] _write: fe fe 3 10 0 51`

While running these same commands with the left arm, the code works perfectly and receives an acknowledgment (or similar) from the port.

```
... _write: fe fe 3 10 0 51
... _read : fe fe 4 10 1 fd b1
```

Could you please advise on how I can get the right arm functional again?

shale-bridge, 2024-09-06:
@afishtex Problem solved by reflashing arm firmware using included scripts on the B1.

closed 2024-09-06 by shale-bridge