elephantrobotics/pymycobot #93: MyCobot280pi stops working. Sometimes with Errormessage, sometimes without.

klotzdafotz, 2024-10-17:
PyMyCobot 3.6.2

When it stops working the only solution is Power off, move all joints of the Cobot and restart, very annoying.
GPIO and ATOM I/O Display is still working.

Sometimes there is no Error Message.
Sometime the Errormessage is

 File "/usr/local/lib/python3.8/dist-packages/pymycobot/mycobot280.py", line 328, in sync_send_coords
    self.send_coords(coords, speed, mode)
  File "/usr/local/lib/python3.8/dist-packages/pymycobot/generate.py", line 202, in send_coords
    self.calibration_parameters(class_name=self.__class__.__name__, coords=coords, speed=speed)
  File "/usr/local/lib/python3.8/dist-packages/pymycobot/error.py", line 564, in calibration_parameters
    check_coords(value, robot_limit, class_name, MyCobot280DataException)
  File "/usr/local/lib/python3.8/dist-packages/pymycobot/error.py", line 97, in check_coords
    raise exception_class(
pymycobot.error.MyCobot280DataException: Has invalid coord value, error on index 2. received -1867 .but angle should be -70 ~ 523.9.

anla-xu, 2024-10-17:
Please check the `send_coords(coords, speed, mode)` function. What is the `coords` you passed in? The error message means that the parameter you passed exceeds the machine's range of motion.

klotzdafotz, 2024-10-22:
Found the Problem.
I use get_angles and get_coords and work with the results, but sometimes the cobot sends totally wrong results.

anla-xu, 2024-10-22:
Please provide the specific values ​​of `get_angles` and `get_coords`. In addition, we need to confirm whether the scale lines of each joint are aligned after sending the angle [0,0,0,0,0,0]

closed 2024-10-23 by klotzdafotz