elephantrobotics/pymycobot #94: get_system_version() -1 ...

klotzdafotz, 2024-10-23:
Sometimes the MyCobot280pi stopps working.
With this testscript it shows me, that there are Errors. Why.
The only solution is to power it off and on with the switch.

```
from pymycobot import MyCobot280

mc = MyCobot280('/dev/ttyAMA0',1000000)
print(mc.get_system_version())
print(mc.get_basic_version())
print(mc.is_controller_connected())
mc.sync_send_angles([0,0,0,0,0,0],100)
```

And the Output is
```
-1
-1
-1
```
pip show pymycobot
(also tried 3.6.3 and earlier)
```
Name: pymycobot
Version: 3.6.4
Summary: Python API for serial communication of MyCobot.
Home-page: https://github.com/elephantrobotics/pymycobot
Author: Elephantrobotics
Author-email: weiquan.xu@elephantrobotics.com
License:
Location: /home/er/.local/lib/python3.8/site-packages
Requires: crc, pyserial, python-can
Required-by:
```

python --version
`Python 3.8.10`

Atom Version
`v7.0`

After restart the output shows
```
7.0
-1
1
```
sometimes it works, sometimes it needs another 1 to 3 power off and on's. When it works it still shows -1 for get_basic_version().
What is the Problem?

anla-xu, 2024-10-25:
The 280 Pi machine cannot use the `get_basic_version` interface, which is used by the 280 M5

klotzdafotz, 2024-10-25:
i know, but that's not the Problem.
The Problem is get_system_version and is_controller_connected and that the 280pi won't work sometimes, and need lots of restarts.

klotzdafotz, 2024-10-30:
Same Problem with pymycobot 3.6.5

anla-xu, 2024-10-31:
What are the specific symptoms of 280Pi not working? Cannot read data? There are a few questions that need to be confirmed. After power is turned on, what color is the light at the end? Are all the joints locked?

klotzdafotz, 2024-11-01:
When I power on the 280Pi the LEDs turn green and they stay green.
The joints are locked.
I don’t get any error messages. The GPIO works and I can control the attached suction pump, but it won’t move until I turn it off, move the arm around and turn it on (sometimes 2 or 3 times Power off/on).
I also redid everything with the myCobot 280 PI ubuntu 20.04 image ( [Downloads - Elephant Robotics](https://www.elephantrobotics.com/en/downloads/) )
and Atom (7.0) and pymycobot (3.6.5) use the latest versions

anla-xu, 2024-11-04:
How do you control the movement of the robot arm? Do you control the angle or the coordinates?

klotzdafotz, 2024-11-04:
I use both. I tried a lot with coords, but sometimes it makes crazy moves, so i switched some from coords to angles

anla-xu, 2024-11-04:
It is normal that some coordinate points cannot be reached. So when you say the machine cannot work, do you mean that the machine does not execute when controlling the coordinate movement?

klotzdafotz, 2024-11-04:
No, it stops moving at some position, which is always different and sometimes after 5 minutes, sometimes after 20 minutes.
And then I can't make him move again. Look at my first post. I tried the same with coords instead of angles, no differenz, because
it says that the controller is not connected any more.

anla-xu, 2024-11-04:
Is it impossible to control it even when using the angle? If it cannot be controlled, please confirm the following:
1. Use the is_power_on interface to find the value returned
2. Record and provide the current angle and coordinate information of the robot
Also, please explain the sentence "because
it says that the controller is not connected any more." How did you come to this conclusion? I don't understand

klotzdafotz, 2024-11-04:
print(mc.is_controller_connected()) returns -1

anla-xu, 2024-11-04:
What about other reading interfaces? For example, get_angles, get_coords, is_power_on

klotzdafotz, 2024-11-25:
```
print(mc.get_system_version())
print(mc.is_controller_connected())
print(mc.get_angles())
print(mc.get_coords())
print(mc.is_power_on())
print(mc.get_angles())
print(mc.get_coords())
print(mc.sync_send_angles([0,0,0,0,0,0],100,5))
print(mc.sync_send_angles([90,0,0,0,0,0],100,5))
```
Output:
```
-1
-1
-1
-1
-1
-1
-1
1
1
```

klotzdafotz, 2024-11-25:
After Power Off and Power on again:
```
6.5
1
[-43.41, -112.58, 20.83, 17.31, 2.98, 169.89]
[161.7, -238.5, 58.7, 164.16, -0.13, 57.11]
1
[-43.41, -112.58, 20.83, 17.31, 2.98, 169.89]
[161.7, -238.5, 58.7, 164.16, -0.13, 57.11]
1
1
```
I switched back to 6.5 because you removed Version 7, but nothing changed

klotzdafotz, 2024-11-25:
Maybe there is a Problem with set_fresh_mode.
I use set_fresh_mode(0). It is slower, but it makes exactly every move I want him to do without time.sleep()
With set_fresh_mode(1) I haven't seen this problem til now.
I need set_fresh_mode(0) specifically when moving down the Z axis with sync_send_coords, because that's the only way it goes down the direct way and doesn't make an arc.

klotzdafotz, 2024-12-10:
No Difference with Atom 6.5 and pymycobot 3.5.3

anla-xu, 2024-12-10:
My understanding is that sometimes after booting up, the robot cannot control or read information, but after restarting, it will be normal. Is that what you mean?

klotzdafotz, 2024-12-10:
Yes, like I said before. Sometimes the robot is stuck, and doesn't move anymore, until I restart. And now sometimes it makes total out of control moves.

anla-xu, 2024-12-16:
We are very sorry, we cannot locate the problem from your description. Now we have released the beta version of atom v7.2, you can try it with this v7.2, pymycobot > v3.6.0

closed 2025-03-17 by anla-xu