elephantrobotics/pymycobot #96: `MyArmm(...).set_power_on()` raises exception, even though it technically did work

apockill, 2024-10-29:
**Describe the bug**
When calling MyArm to enable power, it consistently raises an exception. However, power is in fact enables.

Here's the exception I get:
```python3
Traceback (most recent call last):
  File "/home/alex/acton_ai/acton_ai/scripts/validate_robot.py", line 60, in <module>
    main()
  File "/home/alex/acton_ai/acton_ai/scripts/validate_robot.py", line 19, in main
    mover.set_robot_power_on()
  File "/home/alex/pymycobot/pymycobot/myarmm.py", line 114, in set_robot_power_on
    return self._mesg(ProtocolCode.POWER_ON, has_reply=True)
  File "/home/alex/pymycobot/pymycobot/myarm_api.py", line 70, in _mesg
    if len(res) == 0:
TypeError: object of type 'NoneType' has no len()
```

**To Reproduce**
Steps to reproduce the behavior:
```python3
def main():
    logging.basicConfig(
        level=logging.INFO
    )
    mover = MyArmM(port="/dev/ttyACM0", baudrate="1000000")

    # Example that communication does work
    print("Joints angles", mover.get_joints_angle())
    print("Current power", mover.get_robot_power_status())
    mover.set_robot_power_on()  # <-- this line raises the exception
```

**Expected behavior**
I'd expect power to be enabled, without a Traceback.

**Desktop (please complete the following information):**
 - OS: Pop!_OS 22.04 LTS

anla-xu, 2024-10-31:
What version of pymycobot are you using?

apockill, 2024-10-31:
I tested this on 3.6.5

anla-xu, 2024-11-05:
#103 The problem has been fixed, please install the source code and try again

closed 2024-11-05 by anla-xu