elephantrobotics/pymycobot #97: `MyArmM.set_joints_angle(...)``` is currently not functional

apockill, 2024-10-29:
**Describe the bug**
At the moment, it seems `pymycobot` doesn't yet support controlling servos on `MyArmM`. Please let me know if I'm using this wrong. 

Here's the exception I get, with the following code below.
```python3
Traceback (most recent call last):
  File "/home/alex/acton_ai/acton_ai/scripts/validate_robot.py", line 69, in <module>
    main()
  File "/home/alex/acton_ai/acton_ai/scripts/validate_robot.py", line 22, in main
    mover.set_joint_angle(
  File "/home/alex/pymycobot/pymycobot/myarmm.py", line 20, in set_joint_angle
    self.calibration_parameters(class_name=self.__class__.__name__, joint_id=joint_id, angle=angle, speed=speed)
  File "/home/alex/pymycobot/pymycobot/error.py", line 1269, in calibration_parameters
    limit_info = robot_limit[class_name]
KeyError: 'MyArmM'
```

**To Reproduce**
```python3
def main():
    logging.basicConfig(
        level=logging.INFO
    )
    mover = MyArmM(port="/dev/ttyACM0", baudrate="1000000")
    controller = MyArmC(port="/dev/ttyACM1", baudrate="1000000")

    # Example that communication does work
    print("M angles", mover.get_joints_angle())
    print("C angles", controller.get_joints_angle())

    # Set servo 0 on Mover to match servo 0 on Controller
    mover.set_joint_angle(
        joint_id=0,
        angle=controller.get_joints_angle()[0],
        speed=1
    )
```

**Expected behavior**
I expect the servos to be controllable. I'm currently unable to set any values on any servos due to this bug. 

**Desktop (please complete the following information):**
 - OS: Pop!_OS 22.04 LTS

anla-xu, 2024-10-31:
What version of pymycobot are you using?

apockill, 2024-10-31:
I tested this on 3.6.5

apockill, 2024-11-05:
Any updates? I'd love help being able to move my robot around :)

apockill, 2024-11-05:
This PR allows me to communicate with the arm, but `set_joints_angle` still has no effect. 
https://github.com/elephantrobotics/pymycobot/pull/101

anla-xu, 2024-11-05:
#101 The problem has been fixed, please install the source code and try again

closed 2024-11-05 by anla-xu