elephantrobotics/pymycobot #104: MyArmM `get_joints_min` and `get_joints_max` don't match the values in `calibration_parameters`, leading to `ValueError`s

apockill, 2024-11-05:
**Describe the bug**
In order to control MyArmM using MyArmC, I need to copy joint angles from MyArmC to MyArmM. 

I need to know what the min and max values are, so I can clamp the values to the correct values, before sending them to `set_joint_angle`. However, it's impossible to do so right now because `calibration_parameters` has different values than reported by `get_joints_min` and `get_joints_max`.

This change is introduced in https://github.com/elephantrobotics/pymycobot/pull/103/files#diff-edb1753620e8370d80dd9ab73dff12930bbf15bac171ea5b08ca00af41f248e6R355

Where the PR sets

```python3
"angles_min": [-170, -83, -90, -155, -91, -153, -118],
"angles_max": [170, 83, 84, 153, 88, 153, 2],
```

However, if you call `get_joints_min` and `get_joints_max`, you get:
```python3
mover.get_joints_max()=[172.0, 90.0, 91.0, 148.0, 84.0, 146.0, 0.0]
mover.get_joints_min()=[-168.0, -77.0, -86.0, -159.0, -95.0, -161.0, -118.0]
```

This is using v1.1 firmware on both MyArmC and MyArmM.

anla-xu, 2024-11-07:
Please provide the software version, which can be queried using the following interfaces: `get_robot_firmware_version`, `get_robot_modified_version`

anla-xu, 2024-11-11:
The main control software version seems to be incorrect. Please use `myStudio` to re-burn the latest `picoMain` firmware. The correct `get_robot_modified_version` should be 7