elephantrobotics/pymycobot #111: I can't control the mycobot m5 in ubuntu with 10-20HZ control cmd

Black0Moonlight, 2024-11-29:
**Describe the bug**
I can't control the mycobot m5 in ubuntu with 10-20HZ control cmd

**To Reproduce**
Steps to reproduce the behavior:
I use this method to move and check time delay:
    def move_mycobot(self):
        # self.mc.sync_send_angles(list(np.degrees(self.robot_joint_positions)), 50, 0.1)
        last_time = time.time()
        self.mc.send_radians(list(self.robot_joint_positions), 50)
        dt = time.time()-last_time
        print(dt)

And dt is like this:
1.5237412452697754

1.5245559215545654

1.5268266201019287

Why this huge delay?

**Expected behavior**
In user cook book, it is said that command frequency can be 10-20HZ, and I want to control the move of robot arm as real time as possible.
"
2 关于参数
Q：mycobot280-M5通讯频率是多少呢？
A：10-20Hz。
"
**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Browser chrome
 - Version 3.6.0

anla-xu, 2024-11-29:
Please provide the firmware version you are using. We recommend you to install atom v6.5 and pymycobot < v3.6.0

Black0Moonlight, 2024-11-30:
> Please provide the firmware version you are using. We recommend you to install atom v6.5 and pymycobot < v3.6.0

Thank you. I use firmware with: robot version 7.0.0 and basic version 2.3.0. I also flashed atom v6.5, but I found no difference.

anla-xu, 2024-12-04:
The normal communication frequency should not exceed 100ms. Please test it again using atom v6.5 and pymycobot v3.5.9. There is no problem.

closed 2025-02-10 by anla-xu