elephantrobotics/pymycobot #124: sync_send_coords in Mode 1 only works in set_fresh_mode 0 ?

klotzdafotz, 2025-01-22:
(Systemversion 7.2 / pymycobot 3.7.0)
Is it correct, that Mode 1 (linear) in sync_send_coords only works with set_fresh_mode 0?

in "test 0 1" (freshmode 0 and sync_send_coords 1) it works (you can see that it is going down linear)
in "test 1 1" (freshmode 1 and sync_send_coords 1) it doesn't work (you can see that it is going down in a curve)

`from pymycobot import MyCobot280
import time

#test 0 0
mc = MyCobot280('/dev/ttyAMA0',"1000000")
mc.set_fresh_mode(0)
mc.sync_send_angles([0,0,0,0,0,90],100)
time.sleep(1)
mc.sync_send_angles([0,0,-50,-40,0,90],100)
time.sleep(1)
mc.sync_send_coords([150,-65,150,-180,-1,180],100,0,1)
time.sleep(1)

#test 0 1
mc.sync_send_angles([0,0,0,0,0,90],100)
time.sleep(1)
mc.sync_send_angles([0,0,-50,-40,0,90],100)
time.sleep(1)
mc.sync_send_coords([150,-65,150,-180,-1,180],100,1,1)
time.sleep(1)

#test 1 0
mc.set_fresh_mode(1)
mc.sync_send_angles([0,0,0,0,0,90],50)
time.sleep(1)
mc.sync_send_angles([0,0,-50,-40,0,90],50)
time.sleep(1)
mc.sync_send_coords([150,-65,150,-180,-1,180],50,0,1)
time.sleep(1)

#test 1 1
mc.sync_send_angles([0,0,0,0,0,90],50)
time.sleep(1)
mc.sync_send_angles([0,0,-50,-40,0,90],50)
time.sleep(1)
mc.sync_send_coords([150,-65,150,-180,-1,180],50,1,1)`

klotzdafotz, 2025-01-22:
Shorter code for better understanding,
first and third up and down is looking good (linear),
second and fourth up and down is not looking good (curve)
The speed in Freshmode 1 has been reduced, as the maximum speed is much lower in Freshmode 0.

```
from pymycobot import MyCobot280
import RPi.GPIO as GPIO

mc = MyCobot280('/dev/ttyAMA0',"1000000")
mc.sync_send_coords([150,-65,265,-180,-1,180],100,1,1)
c=0
while c<3:
    #test feshmode 0
    mc.set_fresh_mode(0)
    #mc.sync_send_angles([0,0,-50,-40,0,90],100)
    mc.sync_send_coords([150,-65,120,-180,-1,180],100,1,3)
    mc.sync_send_coords([150,-65,265,-180,-1,180],100,1,3)
    #test feshmode 1
    mc.set_fresh_mode(1)
    mc.sync_send_coords([150,-65,120,-180,-1,180],50,1,3)
    mc.sync_send_coords([150,-65,265,-180,-1,180],50,1,3)
    c+=1
```

anla-xu, 2025-02-10:
When set_fresh_mode is set to 0, motion instructions are executed sequentially, indicating that the firmware is set to execute synchronously, and the sync interface is not required; the sync interface is only a synchronization interface on the Python side.

klotzdafotz, 2025-02-10:
I don't know, what you want me to say with that. The main question is, why the arm moves linear with freshmode 0 and sync_send_coords 1 and in a curve with freshmode 1 and sync_send_coords 1

anla-xu, 2025-02-10:
Only when set_fresh_mode is set to 0 can the movement be in a straight line. This is the product design.

closed 2025-02-19 by anla-xu