elephantrobotics/pymycobot #134: Accuracy issue on the 280 M5

eladoni1, 2025-03-05:
Hi, Currently have the versions 2.30 for the base, and 6.50 for the atom. working on C# but also tried with the python's 280_draw_gcode.py and read both versions of the code to understand if there is an accuracy parameter I've missed.

Currently my problem is as follows - 
Issue 1. Accuracy issue when using the coordinates. The coordinates I am requesting is far off from where I end at (Example - requested by 'set_coords' x,y,z,xr,yr,zr of [160, 160, 160, 0, 0, 0], and when it stops and I use 'get_coords', I received [157.8, 164.3, 156.6, -1.27, 1.11, 1.92], Which is FAR from the accuracy promised, which is 0.5mm)

When I instructed the cobot to move AGAIN to [160, 160, 160, 0, 0, 0], it refused to move. Maybe because it's close by a constant to the target, so it decides to not move (?)

***Note - The angle version of this function, 'set_angles' has more accurate movement. When I realized that, I've searched for a function that transfer coordinates to angles (Function in question is 'solve_inv_kinematics'). Unfortunately, such function is unavailable / returns nothing over the serial port.**

**So my question about this issue is as follows** - Is it expected that the coordinates are MUCH less accurate than the angles? also, Is it expected that the function is not available at this version, or for the M5?
Also - is there any deadzone / error tolerance / acceleration / accuracy parameters at all?

Issue 2. Speed parameter is not consistent.. When I put 100 it's fast, but when I put 5 or 30 or 50 It's the same speed..  So some values are irrelevant and some does work..

**My question is** - is this expected, and how should I debug this? Should I use constant speed for the M5?

Issue 3. Mode = 1 (linear movement) parameter seem to not work.. It moves ever so clunky and has one or more stops along the way to the target.. It sometimes stops before it hits itself..

**Again, question is** - Is it expected ? Should users of the M5 NOT use this parameter?
Thanks.

anla-xu, 2025-03-06:
Is your machine brand new? Please try using the atom v7.2 firmware to see if it will work better.

eladoni1, 2025-03-06:
Thanks for the answer @anla-xu .
As you, and the support instructed - We've updated both the firmware, and both C# library and the Python's library

TL;DR below
{
Which by the way - comparing both versions gave us insight that in Python - we trust the data coming in the serial to be in big endian, while on the C# language, you use 'BitConverter.ToInt16', which tends to be different depending on the architecture of the operator's system.. also, In the C# code, when calling 'GetCoords' / 'GetAngles', when retrieving the data by _serialPort.Read, you take the data starting from the 'Protocol code' instead of 1 byte ahead, to take it from where the data should start..This is not only reducing the accuracy DRASTICALLY, this is also bound to make mistakes. We had to fix that on our own.
} TL;DR - C# version is buggy.

Upgrading from 6.5 to 7.2, issues still persist -
1) the issues of accuracy still persist - 10+mm of error when moving linear (mode = 1 = linear tends to make more errors than the angular mode (mode = 0), but still - 5~mm of error. This is far off from the promised accuracy of 0.5mm.
2) On version 7.2 only, Calibration locks the free movement on the joints, causing calibration to be imposible when using the LCD on the base (The LED turns red and doesn't allow us to calibrate - move any of the joints).

Still would like to know if there are any accuracy, deadzone, tolerance parameters that might be affecting our 10mm offset.

And another question - The myCobot sometimes just stops moving after 'SendCoords', sends a blue RGB value to the LEDs. I only assume it's because it went out of bounds.
My question is - is there anyway to know BEFOREHAND if the position we requested is forbidden? or we must send it with 'SendCoords', and if it stays in the same location, ask for 'get_error_information'?

Thank you,
Elad

anla-xu, 2025-03-10:
There are two different performances for unreachable points, which are distinguished by two different modes:

1. queue mode: send unreachable positions, the robot will move to the limit position and stop, and the LED will turn blue.
2. refresh mode: send unreachable positions, the robot will not execute, and the LED will turn blue.

Errors can be obtained through `get_error_information`, and the interface for setting the mode is `set_fresh_mode`

anla-xu, 2025-03-10:
Setting the I value in the PID value of joint 2/3/4 to 3 can provide accuracy. You can use the interface `set_servo_data`, for example, `set_servo_data(2, 23, 3)`

closed 2025-03-17 by anla-xu