elephantrobotics/pymycobot #155: System Images Not Up To Date

JaredBaileyDuke, 2025-05-07:
The system images listed here have a quite old version of pymycobot. As well, they have an old version of blockly. Both do not support the AGV.

As well, I couldn't get the Ubuntu 20.04 version to work.

https://github.com/elephantrobotics/myAGV-docs/blob/main/MYAGV_PI_2023_EN/8-FilesDownload/8.4-SystemInformation/8.4.1-System_Image.md

anla-xu, 2025-05-08:
This Ubuntu 20.04 image is the latest and has not been updated since then. You can update [myBlockly](https://download-elephantrobotics.oss-cn-shenzhen.aliyuncs.com/software/myblockly/myblockly-arm64-latest.AppImage) and pymycobot in the system by yourself

pymycobot update：
`pip install pymycobot -U`

JaredBaileyDuke, 2025-05-08:
I've tried downloading the Ubuntu 20.04 file multiple times, and then using it. I don't think this file works. You may want to examine it.

I've place the pip install pymycobot -U into a separate issue.

anla-xu, 2025-05-08:
What does it mean that the file does not work? Please describe it clearly.

JaredBaileyDuke, 2025-05-08:
I download the file. Then I unzip the file. Then I place the file on an SD card. Then I transfer the SD card to the pi. Then I turn the pi on. The pi does not boot. I have tried this multiple times. It appears that the image is too small. The Ubuntu 20.04 image is a much smaller size than the Ubuntu 18.04 image. I think the 20.04 image isn't complete as it stands on GitHub.

anla-xu, 2025-05-08:
Could you please provide the parameters of the Raspberry Pi you are using? Is it Pi-4B? How much RAM does it have? 4G or 8G?

JaredBaileyDuke, 2025-05-08:
I am using a Pi-4b with 8GB of RAM

anla-xu, 2025-05-08:
It may not support 8G RAM. Our system is based on 4G. We will confirm it.

anla-xu, 2025-05-15:
Just to confirm, did you disassemble the machine and replace the Raspberry Pi inside? Did you replace the Raspberry Pi with your own 8G RAM?

JaredBaileyDuke, 2025-05-15:
We did. The previous user (we're at a school) plugged in the wrong charger and fried the PCB hat and the Pi. We've replaced both.

anla-xu, 2025-05-23:
We tested the 20.04 system and it was normal on the Raspberry Pi-8G.

[image]

closed 2025-05-23 by anla-xu