elephantrobotics/pymycobot #157: Failure updating pymycobot after cloning the repo

JaredBaileyDuke, 2025-05-08:
I have the AGV, and no other hardware. I am trying to update the pymycobot package for python 3. I clone the repo to my Desktop. I then cd into the repo. I execute pip3 install . in my terminal. I get the following error:

File "<string>", line 1, in <module>
File "/tmp/pip-fk9zsi5i-build/setup.py", line 12, in <module> import pymycobot
File "/tmp/pip-fk9zsi5i-build/pymycobot/__init__.py", line 33, in <module> from pymycobot.mycobotpro630 import Phoenix
File "/tmp/pip-fk9zsi5i-build/pymycobot/mycobotpro360.py", line 14, in <module> import crc
ModuleNotFoundError: No module named 'crc'

JaredBaileyDuke, 2025-05-08:
I was able to install crc and dataclasses. This was tricky as dataclasses is for 3.7+. You may want to update your standard software image, since the current one is python 3.6.

I am encountering additional issues. I will post here shortly.

JaredBaileyDuke, 2025-05-08:
Running sudo pip3 install . from the folder pymycobot (on the Desktop)

Error:
File "<string>", line 1, in <module>
File "/tmp/pip0p2248c_-build/setup.py", line 12, in <module> import pymycobot
File "/tmp/pip0p2248c_-build/pymycobot/__init__.py", line 33, in <module> from pymycobot.mycobotpro630 import Phoenix
File ""/tmp/pip0p2248c_-build/pymycobot/mycobotpro630.py", line 62, in <module> if platform.system() == "Linux" and platform.machine() == "aarch64" and is_debian_os():
File "/tmp/pip0p2248c_-build/pymycobot/mycobotpro630.py", line 35, in is_debian_os ["lsb_release", "-a"], capture_output=True, text=True, check=True
File "usr/lib/python3.6/subprocess.py", line 423, in run with Popen(*popenargs, **kwargs) as process: TypeError: __init__() got an unexpected keyword argument 'capture_output'

----

Command python setup.py egg_info" failed with error code 1 in /tmp/pip-0p2248c_-build/

JaredBaileyDuke, 2025-05-08:
ChatGPT response:
Your Raspberry Pi is still using Python 3.6, and pip3 is tied to it.

But in Python 3.7+, the subprocess.run(..., capture_output=True) argument is valid — in Python 3.6 and earlier, it's not, so it throws this error:

TypeError: __init__() got an unexpected keyword argument 'capture_output'

🧠 Why This Keeps Happening
The pymycobot repo includes support for mycobotpro630, and even though you're not using that robot, it gets imported during install — and crashes due to that one Python 3.7+ feature.

It seems that the Ubuntu 18.04 file on your GitHub repo isn't ready to be upgraded

JaredBaileyDuke, 2025-05-08:
Why would your Ubuntu 18.04 image available on GitHub not include python 3.7+?

JaredBaileyDuke, 2025-05-08:
This process to update was very complicated. This is because:
- Both python 2 and 3 are on the image
  -You should stop using python 2 at this point
- The instructions on your GitHub say pip install, not pip3 install
  - You should update the instructions on your GitHub to be more clear
- You have installed python 3.6 on your Ubuntu 18.04 image, but the pymycobot image requires python 3.7+
  - I had to manually update several python files (methods listed above) in order to get everything to install on python 3.6
  - You should update your image to be python 3.7 or above
- Ultimately, the image should be updated to work on the AGV out of the box for a better user experience

I don't have the AGV moving yet, I will update soon

JaredBaileyDuke, 2025-05-08:
I can now get my code to run using sudo python3 in the bash, however the wheels won't turn.

The file prints everything fine. I am using /dev/ttyAMA0 at 152000 baud rate

anla-xu, 2025-05-22:
The problem of strong dependency on third-party libraries has been solved

closed 2025-05-22 by anla-xu