elephantrobotics/pymycobot #177: Running parallel processes possible with MechArm 270 Pi

mannytan, 2025-06-25:
Is it possible to run parallel processes? Whenever I use send_angle to a specific joint, it waits until the send_angle is completed before moving onto the next one. I've tried to switching fresh_mode which only clears the existing queue.

```
from pymycobot import MechArm270
import time

mc = MechArm270('/dev/ttyAMA0', 1000000)

mc.send_angle(1, 45, 25)
mc.send_angle(1, -45, 25)
mc.send_angle(1, 45, 25)
mc.send_angle(1, -45, 25)

mc.send_angle(2, -45, 50)
mc.send_angle(2, -45, 50)
mc.send_angle(2, -45, 50)
mc.send_angle(2, -45, 50)

```

mannytan, 2025-06-27:
I know `send_angles` affects multiple joints at the same time. What I was trying to solve for is how to fluidly move from one position to another.

Below is some sample code i wrote that continuously moves the arm in a circle. There's a slight pause between movements that I can't remove, regardless of the send_coords speed and angle_speed values.

`
from pymycobot import MechArm270
import time
import math

mc = MechArm270('/dev/ttyAMA0', 1000000)
mc.focus_all_servos()
mc.set_fresh_mode(1)

# ===============================================
print("=" * 40)
print("MechArm270 Status")
print(f"is_power_on: {mc.is_power_on()}")
print(f"is_paused: {mc.is_paused()}")
print(f"is_all_servo_enable: {mc.is_all_servo_enable()}")
print(f"get_fresh_mode: {mc.get_fresh_mode()}")

initial_coords = mc.get_coords()
print(f"initial_coords: {initial_coords}")
print("")
print("=" * 40)

# ===============================================

print("MechArm270 Coords movement")
home_coords = [175, 0, 175, 90, 45, 90]

mc.send_coords(home_coords, 50, 0)
while mc.is_moving():
    time.sleep(0.01)

radius = 25
angle = 0
angle_speed = 0.5

while True:
    x = int(home_coords[0] + radius * math.cos(angle))
    z = int(home_coords[2] + -radius * math.sin(angle))
    angle += angle_speed
    circle_coords = [
        x,
        home_coords[1],
        z,
        home_coords[3], 
        home_coords[4],
        home_coords[5]
    ]
    print(f"coords: {circle_coords}")
    mc.send_coords(circle_coords, 100, 0)
    # while mc.is_moving():
    #     time.sleep(0.01)
`

mannytan, 2025-06-30:
That's weird. I copied your code exactly and my arm is still pausing between moves. 
https://github.com/user-attachments/assets/b387f3fd-b399-4cf9-a7ed-7eb3845a7870

anla-xu, 2025-07-01:
Please provide the atom firmware version number you are using

MagiaYan, 2025-07-01:
Hi,The previous code is based on the version of the M5 machine, on the Raspberry Pi version of the machine I encountered the same problem as you, after troubleshooting found that it was a timeout caused by instruction packet loss, I have turned off the judgment of packet loss in the latest code, and now it can be executed normally
1.Please install the lasted firmware from here [(https://github.com/elephantrobotics/pymycobot)](url)
2.Please Run this code

`from pymycobot import MechArm270
import time
import math

mc = MechArm270('/dev/ttyAMA0', 1000000)
mc.focus_all_servos()
mc.set_fresh_mode(1)

print("=" * 40)
print("MechArm270 Status")
print(f"is_power_on: {mc.is_power_on()}")
print(f"is_paused: {mc.is_paused()}")
print(f"is_all_servo_enable: {mc.is_all_servo_enable()}")
print(f"get_fresh_mode: {mc.get_fresh_mode()}")

initial_coords = mc.get_coords()
print(f"initial_coords: {initial_coords}")
print("")
print("=" * 40)

print("MechArm270 Coords movement")
home_coords = [175, 0, 175, 90, 45, 90]

mc.send_coords(home_coords, 50, 0)
while mc.is_moving():
    time.sleep(0.01)

radius = 25
angle = 0
angle_speed = 0.1  #This is a differential velocity that represents the step distance of a cycle
st = time.time()
while True:
    x = int(home_coords[0] + radius * math.cos(angle))
    z = int(home_coords[2] + -radius * math.sin(angle))
    angle += angle_speed
    circle_coords = [
        x,
        home_coords[1],
        z,
        home_coords[3], 
        home_coords[4],
        home_coords[5]
    ]
    #print(f"coords: {circle_coords}")
    #The motor speed(20) we set should match the planned speed
    mc.send_coords(circle_coords, 20, 0, _async=True) # _async=True  Open-loop control
    print("control time: ", 1000*(time.time() - st)) #This is the actual execution cycle of the program
    st = time.time()
    time.sleep(0.05) #If there is a significant difference between the execution cycle and the latency of the actual machine, please contact us`

mannytan, 2025-07-01:
I updated pymycobot and I ran [MagiaYan](https://github.com/MagiaYan) 's code and it worked! Thank you!

The interesting thing is that I ran your previous code with the new installs and it had the same jittery movement.

I added _async=True (the open-loop control parameter) to your previous example and it started working as expected.

https://github.com/user-attachments/assets/89135feb-5113-4124-93a4-e04830cbd2c8

anla-xu, 2025-07-02:
OK, this involves some knowledge of robot trajectory algorithms. If you are not familiar with robots, it will not be easy to control the robot to achieve the movement effect you want. We are also very happy to help you.

[image]